douglasjunior/react-native-easybluetooth-classic #1: Is this working iOS?

0lide, 2017-06-08:
I'm looking rn library which I can use with Android and iOS, is this supporting iOS?

douglasjunior, 2017-06-08:
Unfortunately this version will not work for classic bluetooth on iOS, as it is bureaucratic on the iPhone, as it depends on the MFI.

I have plans to implement [Bluetooth Low Energy](https://github.com/douglasjunior/react-native-easybluetooth-le). This is from Bluetooth 4.0. I already have the code, but I'm still without a Mac to integrate into React Native.

0lide, 2017-06-08:
I need use device over BT RFCOMM, is it possible with your Bluetooth Low Energy library?

douglasjunior, 2017-06-08:
RFCOMM is serial connection, right? Write bytes on one side and read bytes on the other side? If so, the answer is yes. I created these libraries for this.

0lide, 2017-06-09:
Apparently yes, but device is not low energy device, it works  with Class 2.1 Bluetooth. So I believe this ble library not work?

douglasjunior, 2017-06-09:
No. And there is another problem, if this Bluetooth 2.1 device does not have MFI certification it does not work on iOS.

0lide, 2017-06-09:
Yes it have MFI Certification. Thank you of your answer! I'm new with rn and this bt is small issue now :)

closed 2017-06-09 by 0lide

douglasjunior, 2017-06-09:
You're welcome! React Native is very good, it is very easy to create native modules to access native resources, if you need it will not be difficult to create your solution.