douglasjunior/react-native-easybluetooth-classic #3: Printing images through easybluetooth

costabatista, 2017-07-19:
How can i print an image using this library? I've a base64 encoded string that i wanna sendo to print. i've tried to convert this base64 string into byte array, but i don't have success until now.

how is it possible? What is the correct format to print an image? Is there any way to do it through react-native?

douglasjunior, 2017-07-19:
To print a image you need to convert to an `byte array` following the protocol that the printer expects. The library does not do this for you.

[Here](https://github.com/douglasjunior/AndroidBluetoothLibrary/blob/8c445bc705aacc79efe8c7d04d8f69d017dfd7e0/Sample/src/main/java/com/github/douglasjunior/bluetoothsample/BitmapActivity.java#L101) is an example of this conversion using Java.

douglasjunior, 2017-07-19:
A more powerful algorithm, which is not accepted by all printers, is available [here](https://github.com/douglasjunior/AndroidBluetoothLibrary/blob/8c445bc705aacc79efe8c7d04d8f69d017dfd7e0/Sample/src/main/java/com/github/douglasjunior/bluetoothsample/EscPosHelper.java#L51).

costabatista, 2017-07-20:
I converted the base64 string to Uint8Array in javascript, but when i see the printed ticket i see something like that:

[object Uint8Array]

In the ticket is the only thing that was printed. So, i follow some things that i've seen in stack overflow and then i could print in this format.

What i did wrong?

douglasjunior, 2017-07-20:
You do not need base64. You need to convert the Bitmap to a byte array. This should be done via native code.

For this you need to create a [native module](https://facebook.github.io/react-native/docs/native-modules-android.html) in your app.

To transfer the data from Java to JavaScript, you need to convert the Byte Array to an Int Array.  Because native modules [dont accept byte array](https://facebook.github.io/react-native/docs/native-modules-android.html#argument-types).

Once you have this Int Array in JavaScript, you can send it to the printer with the method :

```javascript
service.writeIntArray(imageIntArray)
.then(...)
.catch(...);
```

closed 2017-11-01 by douglasjunior