douglasjunior/react-native-easybluetooth-classic #5: The development server returned response error code: 500

zhanyulong, 2017-08-28:
[image]

douglasjunior, 2017-08-28:
I still do not know why, but in the newer versions of React Native its starting with `babel-preset-react-native`  version `3.0.0` or `2.0.1`.

These versions do not support some of the features we're used. To solve, just update to version `2.1.0`.

```bash
yarn add babel-preset-react-native@2.1.0
```
Or
```bash
npm install --save babel-preset-react-native@2.1.0
```

zhanyulong, 2017-09-01:
fixed my issue! Thx @douglasjunior !

closed 2017-09-01 by douglasjunior

Antakarana, 2018-04-22:
My app's problem was solved too.
Thanks a lot @douglasjunior !

psobhanlo, 2018-07-27:
Thanks a lot @douglasjunior ! solved
`npm install --save babel-preset-react-native@2.1.0`

maniyarpayal, 2018-08-07:
@psobhanlo  million Thanks! This solves my issue.

komalsakhiya, 2019-12-24:
this not solves mt issue