douglasjunior/react-native-easybluetooth-classic #6: read failed, socket might closed or timeout,

sumesh1993, 2017-08-29:
getting the above error on android 6.0.2 'Moto G play'

i am trying to connect to another android device and writedata

this is the log from android studio
[image]

this is the code i am using 
```
EasyBluetooth.startScan()
          .then((devices) => {
          devices.forEach(device => {
                 if(device.address == "XX:XX:XX:XX:XX"){
                     return this.connectDevice(device);
                 }
          });
       })
       .catch((ex) => {
            console.warn(ex);
          });

EasyBluetooth.connect(device)
          .then(() => {
            console.warn("Connected!");
            this.writeData();
          })
          .catch((ex) => {
            console.warn(ex);
        });
```

douglasjunior, 2017-08-29:
This error was thrown by Android. I'm not sure if this is possible, I've never done anything like that. I always connect from smartphone to other hardware like printers, arduino, etc.

If this is possible, I think the other smartphone should be with bluetooth configured for this. And here you need to specify the UUID.

If you find a way to do this native (Java), we can adapt the library.

sumesh1993, 2017-08-29:
ohk.
I am able to connect with my macbook though and write data.
As a test i am using writeIntArray: [1,2,3,4].
where does this data go ? hw can i view it in my MBP ? any idea ? this is my first time working with bluetooth. @douglasjunior

douglasjunior, 2017-08-29:
On Windows you just need to connect to the Bluetooth serial port with CMD, on Mac should be no different, but I've never tried. Look http://becomethesolution.com/blogs/mac/read-and-send-serial-data-in-mac-os-x-terminal

About the `writeIntArray`, In Java this is converted to bytes. I had to use Int because React Native does not accept bytes.

I think it would be more interesting for you to use the `writeln("Works in React Native!")`. Then you can write a common String.

sumesh1993, 2017-08-29:
Ok i'll give it a shot

closed 2017-11-01 by douglasjunior