douglasjunior/react-native-easybluetooth-classic #8: No Response After Write

elmbrent, 2018-01-25:
Hi,

I've setup your library correctly. and it seems to work great but I don't get back a response from the write

`  presureStart() {
    EasyBluetooth.writeln('MEAS.PRES?')
    .then(() => {
      console.log("Response to come...")
    })
    .catch((ex) => {
      console.warn(ex);
    })
  }`

 I then wait for a response like in example with the listener and receive nothing from device. 

`  onDataRead(data) {
    console.log(data)
  }` 

I know the device does return data as I've built a similar app in node using bllue-serial-port which returns a message once I sent the write message.

```

  var util = require('util');
    var DeviceINQ = require("./lib/device-inquiry.js").DeviceINQ;
    var BluetoothSerialPort = require("./lib/bluetooth-serial-port.js").BluetoothSerialPort;
    var serial = new BluetoothSerialPort();

    serial.on('found', function (address, name) {
        console.log('Found: ' + address + ' with name ' + name);

        serial.findSerialPortChannel(address, function(channel) {
            console.log('Found RFCOMM channel for serial port on ' + name + ': ' + channel);

            console.log(name);
            
            if (name !== 'CPG1500-1A00ANKSMJ7') return;

            serial.connect(address, channel, function() {
                console.log('Connected. Sending data...');

                serial.on('data', function(buffer) {
                    console.log('got data: Presure (', buffer.toString('utf-8').substring(0, 9), ')');
                });

                setInterval(() => {
                    serial.write(Buffer.from('MEAS.PRES?'), (err, count) => {})
                }, 500)
                
            });
        });
    });

```

Any ideas why this is not working, I'm out of ideas. 

Thanks,
Brent

douglasjunior, 2018-01-25:
Have you registered the listener?

```js
componentWillMount() {
    this.onDataReadEvent = EasyBluetooth.addOnDataReadListener(this.onDataRead.bind(this));
}
```

elmbrent, 2018-01-25:
Yes I have done this, nothing returned.

douglasjunior, 2018-01-25:
What you setup in `bufferSize` and `characterDelimiter`.

The `OnDataRead` event is triggered when it found the `characterDelimiter` in the received data, or when buffer is completed.

elmbrent, 2018-01-25:
I think its actually the sending of data as I tried to remotely power down the device and got no result. 

I used your example. 

    const config = {
      "uuid": "00001101-0000-1000-8000-00805f9b34fb",
      "deviceName": "Bluetooth Example Project",
      "bufferSize": 1024,
      "characterDelimiter": "\n"
    }

elmbrent, 2018-01-25:
Dont worry fixed! "characterDelimiter": "\r"

douglasjunior, 2018-01-25:
Cool!

elmbrent, 2018-01-26:
How much would it cost for you to write the IOS version of this?

douglasjunior, 2018-01-26:
Please, contact me by email **nassifrroma[ at ].gmail.com**

closed 2018-02-22 by douglasjunior