douglasjunior/react-native-easybluetooth-classic #9: Unable to connect

JungleCatSW, 2018-01-25:
Hi I am able to do a scan and find devices, but when I try to connect it fails.

The failure print the error from here: 
**promise.reject(new IllegalStateException("Unable to connect to: " + name + " [" + address + "]"));**
line 187 [ react-native-easybluetooth-core/library/src/main/java/io/github/douglasjunior/ReactNativeEasyBluetooth/core/CoreModule.java ](core file)

I am just cut and pasting the UUID from the readme, does this need to be set to a particular value ?
Is there any way I can find out why this is failing, just trying it on a Samsung Galaxy connecting to a Nexus 5. Assume I know nothing!

Thanks

douglasjunior, 2018-01-26:
You need to know what is the UUID of the device you are trying to connect to.

The UUID in README is for the [HC-05](https://www.amazon.com/LeaningTech-HC-05-Module-Pass-Through-Communication/dp/B00INWZRNC) and some thermal printers. 

However, I've never tried connecting using two smartphones. The library was created to serial communication.

JungleCatSW, 2018-01-26:
Thanks for the reply! I was testing it on phones, but i will be using with a serial device.

In the java version you can pass null to get all devices, is there an equivalent for this library ?

douglasjunior, 2018-01-26:
With UUID or not, you are able to scan all devices. But, to connect you need the UUID.

What is the model of your serial device?

JungleCatSW, 2018-01-26:
I managed to the UUID of the device and it connects when I put it in thanks for the help!

closed 2018-01-27 by douglasjunior