douglasjunior/react-native-easybluetooth-classic #10: startScan returns duplicated devices.

benevbright, 2018-03-30:
Hi, thanks for the great library. 

I tried scan with below code. 
```
            (async () => {
                const config = {
                    "deviceName": "XXXXX",
                    "bufferSize": 1024,
                    "characterDelimiter": "\n"
                }
                let res = EasyBluetooth.init(config);
                console.log('[BT init]', res);

                res = await EasyBluetooth.startScan();
                console.log('[BT startScan]', res);
            })();
```

It returns 26 objects but the problem is that they are all same objects. 
I'm not testing in a special place but just in the house. 
The address and name of them are all same.

Any help? 

Thanks.

douglasjunior, 2018-03-30:
This is expected. The library returns what you bluetooth device finds.

If you note, the device is the same, but the NAME and the RSSI can be updated.

So, you need to handle this, if the mac address is the same, just update the NAME and the RSSI values.

benevbright, 2018-04-02:
Thanks very much for your quick answer. 

Yes right, mac address is same but RSSI is being updated as you said. 
It's still kind of odd to me because it's too many that I got 26 same devices at single scanning call.

But I got it. 
Thanks.

douglasjunior, 2018-04-02:
I looked again at your question, and I saw that repeating devices are being returned in `startScan`, right?

I think we can handle this in the library.

Initially I thought you were talking about the `onDeviceFound` event.

benevbright, 2018-04-02:
Right. 

By the way, I will try other methods today too, such as `onDeviceFound`. 
`startScan` was only one I tried so far.

Thanks.

aidooyaw1992, 2018-05-11:
why is it that it can only find laptops and iphones but not android

douglasjunior, 2018-05-11:
Devices returned are found by the Android Bluetooth API itself. The library makes no interference at this point.

@aidooyaw1992, Please create a new issue for your question, do not pollute other issues with unrelated messages.

closed 2018-05-11 by douglasjunior