douglasjunior/react-native-easybluetooth-classic #13: android device version

AtlasMaxima, 2018-04-10:
Does this package only works for Android 4.1 (API 16) ? 
 
i.e. it won't work for Android 6 Marshmallow ?

douglasjunior, 2018-04-10:
Works on Android API 16 or greather.

Note that in Android API 23 or greather you need make sure that location is enabled: 

https://github.com/douglasjunior/react-native-easybluetooth-classic#known-issues

douglasjunior, 2018-04-11:
You can test, but I believe it is both.

aidooyaw1992, 2018-05-11:
Please I'm able to scan for devices, I'm able to find devices but cannot connect, I always get EUNSPECIFIED

stephane-r, 2018-10-01:
Same issue, anyone have other result ?

douglasjunior, 2018-10-01:
I will close this issue because he original question is:

> Does this package only works for Android 4.1 (API 16) ?

So, please, open a new issue for a new question.

closed 2018-10-01 by douglasjunior