douglasjunior/react-native-easybluetooth-classic #14: build application failed due to new ClassicPackage()

AtlasMaxima, 2018-04-11:
The app build failed because `import io.github.douglasjunior.ReactNativeEasyBluetooth.classic does not exist` 
which failed to retrieve  ` new MainReactPackage(), new ClassicPackage()`

I followed the steps under installation. Would I need to put the ClassicPackage inside of my project folder somewhere?

douglasjunior, 2018-04-11:
No, the only steps required are those displayed in the Readme.

you can check if `react-native link` has linked everything correctly, see this example: 

##### In ``android/build.gradle`` add the jitpack repository to your project.

```diff
allprojects {
    repositories {
        mavenLocal()
        jcenter()
+       maven {                                                                                                             
+          url "https://jitpack.io"                                                                                       
+       }                                                                                                                   
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$projectDir/../../node_modules/react-native/android"
        }
    }
}
```

##### In `android/app/build.gradle`, add the easy-bluetooth-classic dependency
```diff
dependencies {
+    compile project(':easy-bluetooth-classic')
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile "com.facebook.react:react-native:+"  // From node_modules
}
```

##### In `android/settings.gradle`
```diff
include ':app'
+ include ':easy-bluetooth-classic'
+ project(':easy-bluetooth-classic').projectDir = new File(rootProject.projectDir, '../node_modules/easy-bluetooth-classic/android')
```

##### In `android/app/src/main/java/com/example/MainApplication.java`, add the `ClassicPackage` dependency.
```diff
import com.facebook.react.ReactActivity;
import com.facebook.react.ReactPackage;
import com.facebook.react.shell.MainReactPackage;
+ import io.github.douglasjunior.ReactNativeEasyBluetooth.classic.ClassicPackage;                                                                              

import java.util.Arrays;
import java.util.List;

public class MainApplication extends Application implements ReactApplication {

    ...
    
    /**
     * A list of packages used by the app. If the app uses additional views
     * or modules besides the default ones, add more packages here.
     */
    @Override
    protected List<ReactPackage> getPackages() {
        return Arrays.<ReactPackage>asList(
            new MainReactPackage(),
+           new ClassicPackage()                                                                                          
        );
    }
}

```

aidooyaw1992, 2018-05-09:
thank you for the help

closed 2018-05-09 by douglasjunior