douglasjunior/react-native-easybluetooth-classic #16: Example

dedisetiaji, 2018-04-26:
Hi, 
Would you like give some example to display discovery bluetooth device.
I used devices.foreach and then put it in new state. But i failed to display the list

aidooyaw1992, 2018-05-11:
im trying to use the connect device but its not working

aidooyaw1992, 2018-05-11:
connect(device){
		console.log(device);
		EasyBluetooth.connect(device)
      .then(() => {
        console.log("Connected!");
      })
      .catch((ex) => {
        console.warn(ex);
      })
	} 

onDeviceFound(device) {
    console.log("onDeviceFound");
    console.log(device);
    var peripherals = this.state.peripherals;
        if (!peripherals.has(device.address)){
          console.warn('Got  peripheral', device);
          peripherals.set(device.address, device);
          this.setState({ peripherals });
        }

  }

`<ListView
			        enableEmptySections={true}
			        dataSource={dataSource}
			        renderRow={(item) => {
			        const color = item.connected ? 'green' : '#fff';
			        return (
			            <TouchableHighlight onPress={() => this.connect(item) }>
			            <View style={[styles.row, {backgroundColor: color}]}>
			                <Text style={{fontSize: 12, textAlign: 'center', color: '#333333', padding: 10}}>{item.name}</Text>
			                <Text style={{fontSize: 8, textAlign: 'center', color: '#333333', padding: 10}}>{item.address}</Text>
			            </View>
			            </TouchableHighlight>
			        );
			        }}
		            />   `

closed 2018-05-11 by douglasjunior