douglasjunior/react-native-easybluetooth-classic #22: [Question] Is possible to disconnect

renanborgez, 2018-09-19:
## Question

Hello,
I started to use your plugin and everything is going very well until now. i just want to wonder if is possible to `disconnect` from a device

i'm trying to call: 
```
EasyBluetooth.disconnect(device)
```

But i'm receiving the following error
```
EasyBluetoothClassic.disconnect got 3 arguments, expected 2
```

douglasjunior, 2018-09-21:
Hi, thanks for reporting.

I'll take a look soon.

renanborgez, 2018-09-21:
I know that there is no documentation for `disconnect` inside your README file
But i'm wondering because it could be very usefull :)

douglasjunior, 2018-09-21:
Sorry, now I got the problem.

You can't pass arguments to the function `disconnect`.

https://github.com/douglasjunior/react-native-easybluetooth-classic/blob/8971d151920f4f4749863f271c957ca35c26d40d/android/src/main/java/io/github/douglasjunior/ReactNativeEasyBluetooth/classic/ClassicModule.java#L55-L59

renanborgez, 2018-09-21:
May i open a pull request to put it on your documentation?

douglasjunior, 2018-09-21:
Sure, thanks!

renanborgez, 2018-09-21:
:)

closed 2018-09-21 by renanborgez