douglasjunior/react-native-easybluetooth-classic #23: Can't connect to any device

stephane-r, 2018-10-01:
Hi community,

I try to connect my `device` detected with `startScan` method, but i've an error : 

`{ [Error: Unable to connect to: Galaxy Tab A (2016) [F0:EE:10:C5:56:6F]] framesToPop: 1, code: 'EUNSPECIFIED' }`

I've try other device, but same result. I use same `config` than on README. Maybe i need other uuid ?

I use Huawai P20 on Android 8.

bchaves2604, 2019-01-28:
same problem here

bchaves2604, 2019-01-28:
I found the solution, you must pass the UUID showed in the config var

const config = {
      "uuid": "00001101-0000-1000-8000-00805f9b34fb",
      "deviceName": "OBDII",
      "bufferSize": 1024,
      "characterDelimiter": "\n"
    }

douglasjunior, 2019-01-28:
Hello guys, note that the UUID is particular to the target device you want to connect to, the README is just a example.

If you try to connect to a thermal printer, then you need to put the UUID of the thermal printer bluetooth device.

If you try to connect to a arduino (as HC-05), then you need to put the UUID of the HC-05 device.

bchaves2604, 2019-01-28:
Thanks for the explanation, probably it worked for that reason

daniyal-intellicel, 2019-04-29:
@bchaves2604 kindly tell me where should I pass it and what should I pass?