douglasjunior/react-native-easybluetooth-classic #24: issue on ios part

nayrskei01, 2018-10-11:
we are using this on a react native but it crash on ios part since it is only for android.
so our work around is to configure the index.js part
we add this line of codes. thanks
import { NativeModules, DeviceEventEmitter, Platform } from 'react-native';

const EasyBluetooth = undefined;
if (Platform.OS === 'android') {
    EasyBluetooth = NativeModules.EasyBluetoothClassic;

    EasyBluetooth.addOnDeviceFoundListener = function (eventCallback) {
        return DeviceEventEmitter.addListener(EasyBluetooth.EVENT_DEVICE_FOUND, eventCallback);
    }

    EasyBluetooth.addOnDataReadListener = function (eventCallback) {
        return DeviceEventEmitter.addListener(EasyBluetooth.EVENT_DATA_READ, eventCallback);
    }

    EasyBluetooth.addOnStatusChangeListener = function (eventCallback) {
        return DeviceEventEmitter.addListener(EasyBluetooth.EVENT_STATUS_CHANGE, eventCallback);
    }

    EasyBluetooth.addOnDeviceNameListener = function (eventCallback) {
        return DeviceEventEmitter.addListener(EasyBluetooth.EVENT_DEVICE_NAME, eventCallback);
    }
}

module.exports = EasyBluetooth;

douglasjunior, 2018-10-11:
You're right, I'll look into this soon.

tonmanayo, 2018-11-16:
@douglasjunior will this ever support IOS?

DaBs, 2018-12-04:
@tonmanayo The issue is that to qualify for using Bluetooth Classic on iOS, you'll need an MFi certified product and you'd be using the EAAccessory protocol instead, which really has very little to do with Bluetooth, apart from it being one of the methods of connecting to such a device.

douglasjunior, 2018-12-04:
@DaBs is right.

nayrskei01, 2018-12-05:
this will not support ios.. and it only uses android,
you need to filter the os platform for the next time the user will install
it it will not crash in ios.