douglasjunior/react-native-easybluetooth-classic #28: How to read data?

bchaves2604, 2019-01-28:
I have the write and read methods  and it seems to work as  #expected.

EasyBluetooth.writeln("Works in React Native!")
                      .then(() => {
                        console.log("Writing...")
                      })
                      .catch((ex) => {
                        console.warn(ex);
                      })

But I want to retrieve data from the device which is and OBD II, any ideas?

douglasjunior, 2019-01-29:
You need to use the [`onDataReadListener`](https://github.com/douglasjunior/react-native-easybluetooth-classic#events).

bchaves2604, 2019-01-29:
I have it but when I log data it shows in the console "Works in React Native!" and what I actually want to do is receive something form the device

onDataRead(data) {
    console.log("onDataRead", data);
  }

componentWillMount() {
    this.onDeviceFoundEvent = EasyBluetooth.addOnDeviceFoundListener(this.onDeviceFound.bind(this));
    this.onStatusChangeEvent = EasyBluetooth.addOnStatusChangeListener(this.onStatusChange.bind(this));
    this.onDataReadEvent = EasyBluetooth.addOnDataReadListener(this.onDataRead.bind(this));
    this.onDeviceNameEvent = EasyBluetooth.addOnDeviceNameListener(this.onDeviceName.bind(this));
  }