douglasjunior/react-native-easybluetooth-classic #30: Impossible to read data

manelsanz, 2019-02-14:
I tried with this package and 'react-native-bluetooth-serial-next' and it's impossible to read nothing.
I can write data to my pc throught a virtual COM Bluetooth incoming port, but no luck with read. I'm using Tera term for Windows for read and write from PC.
Instead, if I try the Serial Bluetooth Terminal app (https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=es) I have no problems to write and read instantly using the same setup.

I need help, that's very important... thank you.

manelsanz, 2019-02-15:
Now works... was the damned delimiter! Was \r instead of \n.... Why we need a delimiter? What I can do if I want all string without cut it!? Put "" as delimiter don't work, thanks.

douglasjunior, 2019-02-15:
You should set a delimiter or define a small buffer size.

These are the two ways to tell to the library when you want to receive the data.

taj567, 2022-01-04:
import React from 'react';
import {
  Platform,
  ScrollView,
  Switch,
  Text,
  SafeAreaView,
  View,
  ActivityIndicator,
  Modal,
} from 'react-native';

import Toast from '@remobile/react-native-toast';
import BluetoothSerial, {
  withSubscription,
} from 'react-native-bluetooth-serial-next';
import {Buffer} from 'buffer';

import Button from './components/Button';
import DeviceList from './components/DeviceList';
import styles from './styles';

global.Buffer = Buffer;

const iconv = require('iconv-lite');

class App extends React.Component {
  constructor(props) {
    super(props);
    this.events = null;
    this.state = {
      isEnabled: false,
      device: null,
      devices: [],
      scanning: false,
      processing: false,
    };
  }

  async componentDidMount() {
    this.events = this.props.events;

    try {
      const [isEnabled, devices] = await Promise.all([
        BluetoothSerial.isEnabled(),
        BluetoothSerial.list(),
      ]);

      this.setState({
        isEnabled,
        devices: devices.map(device => ({
          ...device,
          paired: true,
          connected: false,
        })),
      });
    } catch (e) {
      Toast.showShortBottom(e.message);
    }

    this.events.on('bluetoothEnabled', () => {
      Toast.showShortBottom('Bluetooth enabled');
      this.setState({isEnabled: true});
    });

    this.events.on('bluetoothDisabled', () => {
      Toast.showShortBottom('Bluetooth disabled');
      this.setState({isEnabled: false});
    });

[... 471 lines cut ...]
            />
          </React.Fragment>
        )}

        <View style={styles.footer}>
          <ScrollView horizontal contentContainerStyle={styles.fixedFooter}>
            {isEnabled && (
              <Button
                title="Discover more"
                onPress={this.discoverUnpairedDevices}
              />
            )}
            {!isEnabled && (
              <Button title="Request enable" onPress={this.requestEnable} />
            )}
          </ScrollView>
        </View>
      </SafeAreaView>
    );
  }
}

export default withSubscription({subscriptionName: 'events'})(App);

---how can we get the device data when it is connected through the bluetooth---