douglasjunior/react-native-easybluetooth-classic #36: Init Failure on Prod, Dev Working.

zocheyado, 2019-10-08:
Super weird issue I am running into. When I run a debug build of my app everything about this package works great. However when I generate a signed APK for release it fails. I wrapped it in a try catch and this is the error I am getting on init. 

```
'Error!', { [Error: length=0; index=0]
  framesToPop: 1,
  nativeStackAndroid:
   [ { methodName: 'a',
       lineNumber: 188,
       file: 'BluetoothService.java' },
     { methodName: 'init', lineNumber: 115, file: 'CoreModule.java' },
     { methodName: 'init',
       lineNumber: 34,
       file: 'ClassicModule.java' },
     { methodName: 'invoke', lineNumber: -2, file: 'Method.java' },
     { methodName: 'invoke',
       lineNumber: 372,
       file: 'JavaMethodWrapper.java' },
     { methodName: 'invoke',
       lineNumber: 158,
       file: 'JavaModuleWrapper.java' },
     { methodName: 'run',
       lineNumber: -2,
       file: 'NativeRunnable.java' },
     { methodName: 'handleCallback',
       lineNumber: 883,
       file: 'Handler.java' },
     { methodName: 'dispatchMessage',
       lineNumber: 100,
       file: 'Handler.java' },
     { methodName: 'dispatchMessage',
       lineNumber: 29,
       file: 'MessageQueueThreadHandler.java' } ],
  userInfo: null,
  code: 'EUNSPECIFIED',
  line: 22,
  column: 1799,
```

Everything works perfectly if I debug, the only way I found this is by putting console logs and a try catch into my production build, and then watching ADB logcat. Any ideas?
I know the package is being imported because just before the try catch that throws that, I console log EasyBluetooth and it spits out all of the helpers / functions like you would expect.

douglasjunior, 2019-10-08:
Are you using proguard?

zocheyado, 2019-10-08:
I saw your other thread about a similar issue and I looked. I do not believe I am using progaurd. Started as a simple expo project, ejected and used react link, added all my logic. So unless progaurd is enabled by default I should not have it. I also tried disabling progaurd just incase and I had no luck.

zocheyado, 2019-10-08:
I take it all back! I did a global search in my android studio for proguard and found this
```
  buildTypes {
    debug {
      debuggable true
      ext.enableCrashlytics = false
    }
    release {
      minifyEnabled true
      proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
      consumerProguardFiles 'proguard-rules.pro'
      signingConfig signingConfigs.release
    }
  }
```
I removed both proguard lines and made minifyEnable: false and it works! Thanks for the help. Might want to put that on the readme to save you some headache, people coming form Expo probably run into this without knowing.

closed 2019-10-08 by zocheyado